Arduino-IRremote/Arduino-IRremote #1183: [Bug]: Send and receive doesnt work in attiny167

freskpe, 2023-10-21:
### The type of board you are using.

Other - please specify below

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

attiny167(AttinyCore by spence konde)

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Other - please specify below

### Pin(s) used for IR-receive, if not default.

Receiver PIN 14 and SENDER 13

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [X] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [X] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

#include <Arduino.h>

/*
 * Specify DistanceWidthProtocol for decoding. This must be done before the #include <IRremote.hpp>
 */
#define DECODE_DISTANCE_WIDTH // Universal decoder for pulse distance width protocols

#define IR_RECEIVE_PIN 11
#define IR_SEND_PIN 13
#define SEND_PWM_BY_TIMER 
#define IR_USE_AVR_TIMER1
#define NO_LED_FEEDBACK_CODE // saves 92 bytes program memory
#define RAW_BUFFER_LENGTH 50

//

//#define NO_LED_FEEDBACK_CODE      // saves 92 bytes program memory
//#define RECORD_GAP_MICROS 12000   // Default is 5000. Activate it for some LG air conditioner protocols
//#define SEND_PWM_BY_TIMER         // Disable carrier PWM generation in software and use (restricted) hardware PWM.
//#define USE_NO_SEND_PWM           // Use no carrier PWM, just simulate an active low receiver signal. Overrides SEND_PWM_BY_TIMER definition

//#define DEBUG // Activate this for lots of lovely debug output from the decoders.

#include <IRremote.hpp>

#define SEND_BUTTON_PIN 12

#define DELAY_BETWEEN_REPEATS_MILLIS 50

// Storage for the recorded code, pre-filled with NEC data
IRRawDataType sDecodedRawDataArray[RAW_DATA_ARRAY_SIZE] = { 0x7B34ED12 }; // address 0x12 command 0x34
DistanceWidthTimingInfoStruct sDistanceWidthTimingInfo = { 9000, 4500, 560, 1690, 560, 560 }; // NEC timing
uint8_t sNumberOfBits = 32;

bool sSendButtonWasActive;

void setup() {
    pinMode(SEND_BUTTON_PIN, INPUT_PULLUP);
    ADCSRA = 0;

    // Start the receiver and if not 3. parameter specified, take LED_BUILTIN pin from the internal boards definition as default feedback LED
    IrReceiver.begin(IR_RECEIVE_PIN, DISABLE_LED_FEEDBACK, 6);
    IrSender.begin(); // Start with IR_SEND_PIN as send pin and enable feedback LED at default feedback LED pin
    }

void loop() {

    // If button pressed, send the code.
    bool tSendButtonIsActive = (digitalRead(SEND_BUTTON_PIN) == LOW); // Button pin is active LOW

    /*
     * Check for current button state
     */
    if (tSendButtonIsActive) {
        if (!sSendButtonWasActive) {
          //  IrReceiver.stop();
        }
        /*
         * Button pressed -> send stored data
         */

        IrSender.sendPulseDistanceWidthFromArray(38, &sDistanceWidthTimingInfo, &sDecodedRawDataArray[0], sNumberOfBits,
#if defined(USE_MSB_DECODING_FOR_DISTANCE_DECODER)
                PROTOCOL_IS_MSB_FIRST
#else
                PROTOCOL_IS_LSB_FIRST
#endif
                , 100, 0);

        delay(DELAY_BETWEEN_REPEATS_MILLIS); // Wait a bit between retransmissions

    } else if (sSendButtonWasActive) {
        /*
         * Button is just released -> activate receiving
         */
        // Restart receiver
        IrReceiver.start();

    } else if (IrReceiver.decode()) {
        /*
         * Button is not pressed and data available -> store received data and resume
         * DistanceWidthTimingInfo and sNumberOfBits should be constant for all keys of the same IR remote / protocol
         */
        if (IrReceiver.decodedIRData.protocol != UNKNOWN) {

            if (memcmp(&sDistanceWidthTimingInfo, &IrReceiver.decodedIRData.DistanceWidthTimingInfo,
                    sizeof(sDistanceWidthTimingInfo)) != 0) {
                IrReceiver.printDistanceWidthTimingInfo(&Serial, &IrReceiver.decodedIRData.DistanceWidthTimingInfo);
                sDistanceWidthTimingInfo = IrReceiver.decodedIRData.DistanceWidthTimingInfo; // copy content here
            } else {
            }
            if (sNumberOfBits != IrReceiver.decodedIRData.numberOfBits) {
                sNumberOfBits = IrReceiver.decodedIRData.numberOfBits;
            }
            if (sDecodedRawDataArray[0] != IrReceiver.decodedIRData.decodedRawDataArray[0]) {
                *sDecodedRawDataArray = *IrReceiver.decodedIRData.decodedRawDataArray; // copy content here
            }
        }
       IrReceiver.resume(); // resume receiver
    }

    sSendButtonWasActive = tSendButtonIsActive;
    delay(100);
}

### What happens?

when i comment the line "IrReceiver.begin(IR_RECEIVE_PIN, DISABLE_LED_FEEDBACK, 6);" and pulse the button, the attiny send the code recorded manully in the memory... but if uncomment the line "IrReceiver.begin(IR_RECEIVE_PIN, DISABLE_LED_FEEDBACK, 6);" the attiny167 stop working. 

### The serial output which indicates the error happened.

```shell
no serial interface...

i test another pins for receiver and dont get good results.
```

### What were you expecting to happen?

the attiny send the new code recorded from receiver. 

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

freskpe, 2023-10-21:
i was reading the code in IRTimer.hpp, i note that **timerEnableReceiveInterrupt** and **timerConfigForReceive** functions work with A register in the timer but i'm using the timer B (#define SEND_PWM_BY_TIMER
#define IR_USE_AVR_TIMER1) 

i'm not a expert in microcontrollers, sorry in advance.

ArminJo, 2023-10-21:
For me it looks like you modified the library somewat (but it seems to be a secret) and now want to have support.

closed 2023-10-21 by ArminJo

freskpe, 2023-10-21:
> For me it looks like you modified the library somewat (but it seems to be a secret) and now want to have support.

I only change the pin of timer in IrTimer.hpp because i use the PB6 for sender not PA2. 

I dont have knowledge to change the library... And in the worst case i download and install the library again.

freskpe, 2023-10-29:
i solve the problem, changing the configuration of the timer, in my case, when SEND_PWM_BY_TIMER
and IR_USE_AVR_TIMER1 is needed to change the configuration of the receiver timer to the chanel B of the timer.