Arduino-IRremote/Arduino-IRremote #1194: Cannot change IR_SEND_PIN dynamically

alazifi, 2023-12-30:
### The type of board you are using.

ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

ESP32 DEVKIT V1

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [X] SimpleReceiver
- [X] ReceiveDemo
- [X] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [X] SimpleSender
- [X] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [X] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [X] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

I have a remote multiple emitter cloning project on esp32. The first emitter is on pin 18 and the second emitter is on pin 27 with the code as follows

`String msg;

const char* prontoData() {
  const char* command = nullptr;
  const char* command1[] PROGMEM = "0000 0067 ... ...";
  const char* command2[] PROGMEM = "0000 0067 ... ...";

  if (msg == "A1") {
    command = commmand1;
  } else if (msg == "A2") {
    command = command2;
  }
  return command;
}

void sendData() {
  const int IR_PIN_1 = 18;
  const int IR_PIN_2 = 27;
  int irPin = 0;
  
  String incomingMsg;
  if (msg == "A1") {
    irPin = IR_PIN_1;
  } else if (msg == "A2") {
    irPin = IR_PIN_2;
  }

  IrSender.setSendPin(irPin);
  irsend.sendPronto(F(prontoData()), 3U);
}`

I try to only connected the emitter on pin 27, so that when "A1" was triggered no data would be received by the receiver because the emitter was not connected to pin 18. Next I triggered "A2" so that the emitter on pin 27 could send data and the receiver was also detected Pronto data is sent because the emitter is actually connected to pin 27. 

### What happens?

The problem is that when I continue this step by giving the message "A1" there should be no message belonging to A1 detected by the receiver, but the receiver detects a message belonging to A1 sent via pin 27.

### What were you expecting to happen?

My expectation with this coding is that when there is an incoming message "A1" then the emitter on pin 18 sends data belonging to A1 and if the message comes in "A2" then the emitter on pin 27 sends data belonging to A2.

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2023-12-30:
Please add
`ledcDetachPin(IrSender.sendPin);  // unbind pin`
after 
`irsend.sendPronto(F(prontoData()), 3U);`
and check again.

alazifi, 2023-12-31:
Great. It worked as expected!

ArminJo, 2023-12-31:
Thanks for testing!

commit 401ccb94 (2023-12-31): Fixes #1194
commit 1144c991 (2023-12-31): Fixes #1194
closed 2023-12-31 by ArminJo