Arduino-IRremote/Arduino-IRremote #1203: [Bug]: SimpleSender cannot compile IrSender code instruction copied from ReceiveDemo output from a Mitsubishi IR remote

mackrasp, 2024-02-15:
### The type of board you are using.

Other - please specify below

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

Arduino Nano 33 IOT

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Unknown

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [X] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [X] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1 Upload ReceiveDemo.ino to Nano V3 on IDE 2.3.0
2 Use Mitsubishi AC remote
3 Copy Serial Monitor output and paste into SimpleSender.inoBoard type
4 Edit SimpleSender.ino so that the new code from ReceiveDemo.ino will be active.
5 Attempt to Verify edited code using a Nano 33 IOT.
6 Errors as below

### What happens?
[... 55 lines cut ...]
Using library IRremote at version 4.2.1 in folder: /Users/mack/Documents/Arduino/libraries/IRremote 
exit status 1

Compilation error: no matching function for call to 'IRsend::sendPulseDistanceWidthFromArray(int, int, int, int, int, int, int, uint32_t*, int, int, int, int)'
```

### What were you expecting to happen?

I expected a successful compilation with the Nano 33 IOT.
The Nano 33 IOT also produces the correct NEC IR output when running SimpleSender.ino after uploading from the IDE 2.3.0
The hardware both Nano 33 IOT and Nano V3 do not exhibit any issues otherwise eg they run Blink, Their digital pins can be set High or LOW.

### Additional context.

If I attempt to Verify the ReceiveDemo.ino code using basic Nano (V3?) it verify's without error.
However I wish to use a Nano 33 IOT as it has a WIFI capability.

I note that the Nano 33 IOT uses a SAMD processor but is not shown in the top check box for Board type on this form and although the PinDefinitionsAndMore.h says Pin 4 is the IR output for SAMD it is D3 which is the actual IR output working after uploading which is the AVR/Nano platform!

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

mackrasp, 2024-02-15:
A small correction to my Bug
The on Serial output from ReceiveDemo.ino is 
IrSender.sendPulseDistanceWidthFromArray(38, 3400, 1600, 450, 1200, 450, 350, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);

The code i have copied and pasted into SimplerSender.ino is
IrSender.sendPulseDistanceWidthFromArray(38, 3400, 1600, 450, 1200, 450, 350, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, 0, 0);

ArminJo, 2024-02-15:
If you have enabled `DECODE_DISTANCE_WIDTH`, the code printed by `printIRSendUsage()` **differs between 8 and 32 bit platforms**, so it is best to run the receiving program on the same platform as the sending program.

closed 2024-02-15 by ArminJo