Arduino-IRremote/Arduino-IRremote #1214: [Bug]: last two bytes incorrect of capture of Mitsubishi HVAC IR

chaseadam, 2024-03-14:
### The type of board you are using.

ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Other - please specify below

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. Attach IR receiver to default pin for esp32 (and separately ESP32c3)
2. upload ReceiveDemo
3. press power button on mitsubishi remote
4. IR code captured is correct for first 2 elements of the array, but incorrect for last item

### What happens?

Protocol=PulseDistance Raw-Data=0xCB23 144 bits LSB first
Send on a 32 bit platform with:
    uint64_t tRawData[]={0x150800000126CB23, 0x594030, 0xCB23};
    IrSender.sendPulseDistanceWidthFromArray(38, 3500, 1650, 500, 1250, 500, 400, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);

I believe this protocol repeats once with a large gap and the last two bytes appear to be from the "repeat" as it is the first two bytes of the header for this protocol

I have tried with the following as well as the default 

```
#define RECORD_GAP_MICROS 16000
```

### What were you expecting to happen?

I was expecting a 0x00 byte in addition to a CRC byte

### Additional context.

this is the most succinct description of the protocol I have found: https://github.com/esphome/esphome/blob/d3842a7ab49c32813185d00e4b6d8399c2b1dad7/esphome/components/mitsubishi/mitsubishi.cpp#L92

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

chaseadam, 2024-03-14:
it is notable that the IRremoteESP8266 does capture this correctly (but it is also "aware" of this protocol). That may disqualify this bug, but I am curious why this library gets it mostly right, but not quite...

```
15:21:25.410 -> Timestamp : 000035.469
15:21:25.410 -> Library   : v2.8.6
15:21:25.410 -> 
15:21:25.410 -> Protocol  : MITSUBISHI_AC
15:21:25.410 -> Code      : 0x23CB26010000081530405B000000000000FD (144 Bits)
15:21:25.410 -> Mesg Desc.: Power: Off, Mode: 1 (Heat), Temp: 21.5C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): 3 (Middle), Clock: 15:10, On Timer: 00:00, Off Timer: 00:00, Timer: -, Weekly Timer: Off, 10C Heat: Off, ISee: Off, Econo: Off, Absense detect: Off, Direct / Indirect Mode: 0, Fresh: Off
15:21:25.476 -> uint16_t rawData[583] = {3502, 1672,  514, 1214,  492, 1230,  478, 392,  472, 394,  518, 356,  472, 1252,  498, 374,  486, 382,  492, 1230,  472, 1256,  488, 376,  498, 1236,  486, 382,  490, 376,  474, 1250,  492, 1238,  490, 378,  496, 1230,  476, 1250,  494, 376,  472, 396,  496, 1236,  494, 372,  494, 372,  478, 1250,  476, 390,  492, 376,  498, 372,  514, 356,  472, 392,  470, 400,  472, 392,  526, 358,  514, 350,  498, 376,  476, 388,  496, 368,  514, 356,  470, 394,  476, 396,  496, 368,  476, 394,  470, 394,  498, 368,  494, 378,  490, 380,  514, 350,  494, 376,  496, 372,  470, 390,  496, 378,  512, 1214,  474, 398,  472, 388,  476, 394,  472, 394,  488, 1234,  476, 394,  472, 1254,  490, 384,  496, 1234,  488, 382,  472, 396,  472, 396,  498, 370,  492, 378,  492, 372,  494, 376,  472, 1252,  494, 1238,  498, 376,  474, 394,  472, 392,  496, 372,  472, 396,  476, 392,  496, 372,  472, 394,  470, 1260,  486, 380,  492, 1236,  492, 1234,  496, 372,  514, 1214,  488, 1240,  486, 378,  496, 1232,  490, 378,  490, 380,  494, 376,  488, 380,  494, 376,  492, 376,  492, 380,  494, 378,  476, 394,  514, 352,  498, 372,  470, 398,  494, 378,  496, 366,  498, 370,  476, 402,  488, 376,  496, 372,  496, 376,  496, 368,  490, 380,  490, 378,  514, 356,  496, 376,  470, 398,  490, 380,  492, 382,  496, 368,  494, 378,  472, 392,  498, 368,  492, 376,  472, 394,  498, 368,  472, 394,  494, 374,  478, 388,  498, 378,  492, 382,  470, 394,  490, 376,  476, 392,  472, 396,  512, 358,  516, 352,  520, 350,  492, 372,  498, 372,  472, 396,  472, 1256,  514, 352,  496, 1230,  492, 1240,  488, 1240,  496, 1226,  494, 1234,  492, 1238,  494, 17104,  3648, 1662,  492, 1234,  492, 1238,  498, 368,  494, 370,  498, 372,  518, 1212,  474, 410,  488, 376,  472, 1252,  496, 1234,  498, 368,  512, 1214,  514, 356,  526, 352,  476, 1250,  478, 1248,  490, 380,  486, 1240,  488, 1240,  494, 370,  498, 370,  488, 1246,  498, 370,  470, 396,  488, 1240,  492, 376,  494, 372,  492, 376,  478, 394,  494, 376,  490, 378,  498, 368,  492, 378,  490, 378,  494, 376,  488, 376,  472, 396,  488, 376,  496, 378,  494, 380,  488, 382,  488, 382,  490, 380,  496, 368,  472, 394,  498, 370,  490, 376,  498, 370,  498, 368,  494, 380,  476, 392,  522, 1212,  472, 398,  492, 378,  496, 368,  594, 276,  490, 1234,  498, 374,  472, 1250,  496, 372,  488, 1240,  494, 376,  478, 392,  472, 394,  488, 380,  498, 372,  492, 382,  486, 390,  492, 1236,  490, 1238,  472, 394,  476, 394,  496, 378,  490, 372,  494, 376,  470, 394,  476, 392,  490, 378,  476, 1252,  498, 374,  500, 1238,  490, 1238,  470, 394,  476, 1252,  498, 1226,  494, 374,  498, 1234,  472, 394,  476, 392,  470, 394,  474, 392,  496, 370,  496, 372,  490, 380,  472, 394,  488, 380,  490, 376,  492, 380,  470, 394,  476, 388,  496, 374,  522, 356,  472, 392,  492, 392,  490, 378,  490, 376,  476, 388,  498, 374,  486, 382,  496, 368,  498, 370,  472, 394,  514, 358,  494, 374,  492, 368,  498, 376,  494, 380,  494, 384,  468, 392,  514, 350,  492, 376,  472, 394,  498, 374,  476, 388,  518, 356,  494, 376,  496, 368,  472, 392,  498, 372,  476, 390,  492, 378,  496, 370,  498, 366,  476, 390,  476, 402,  490, 374,  496, 1232,  470, 394,  492, 1240,  490, 1234,  494, 1234,  490, 1242,  512, 1214,  500, 1230,  490};  // MITSUBISHI_AC
15:21:25.740 -> uint8_t state[18] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x00, 0x08, 0x15, 0x30, 0x40, 0x5B, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFD};
15:21:25.774 -> 
```

ArminJo, 2024-03-14:
With 144 bits you do not get rawData[583]. 
Can you connect the debug pin to ground for ReceiveDemo, so we can see the raw data of IRremote?

ArminJo, 2024-03-14:
The look quite similar
```
0x23CB260100000815    30405B0000000000 00FD   - IRremote
0x150800000126CB23, 0x594030,        0xCB23   - IRremoteESP8266 
```

chaseadam, 2024-03-15:
My understanding with LSB that they are nearly identical, but the last two bytes are always the same as the first two bytes (and I believe this IR code repeats one time)

here is the timing output requested with the "on" and "off" signal from the remote. Seems like "36" may be too short for 144 bits, but I am not sure how they convert. 

A quick compare to the rawData[583] of IRremoteESP82666 looks like it has less precision (and is much shorter). It came from the 'IRrecvDump V2' example.

```
21:11:18.301 -> Protocol=PulseDistance Raw-Data=0xCB23 144 bits LSB first
21:11:18.301 -> Send on a 32 bit platform with: 
21:11:18.301 ->     uint64_t tRawData[]={0x150800000126CB23, 0x7E4030, 0xCB23};
21:11:18.301 ->     IrSender.sendPulseDistanceWidthFromArray(38, 3550, 1650, 450, 1250, 450, 400, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);
21:11:19.229 -> 
21:11:19.229 -> Protocol=PulseDistance Raw-Data=0xCB23 144 bits LSB first
21:11:19.229 -> Send on a 32 bit platform with: 
21:11:19.229 ->     uint64_t tRawData[]={0x150820000126CB23, 0x7E4030, 0xCB23};
21:11:19.229 ->     IrSender.sendPulseDistanceWidthFromArray(38, 3550, 1600, 450, 1250, 450, 400, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);
21:11:26.451 -> 
21:11:26.451 -> rawData[36]: 
21:11:26.451 ->  -3276750
21:11:26.451 ->  +3500,-1650
21:11:26.451 ->  + 500,-1250 + 500,-1200 + 550,- 350 + 500,- 350
21:11:26.451 ->  + 550,- 350 + 500,-1200 + 500,- 400 + 500,- 350
21:11:26.484 ->  + 500,-1200 + 500,-1250 + 500,- 350 + 500,-1250
21:11:26.484 ->  + 500,- 350 + 500,- 400 + 500,-1200 + 500,-1250
21:11:26.484 ->  + 500
21:11:26.484 -> Sum: 26450
21:11:27.643 -> 
21:11:27.643 -> rawData[36]: 
21:11:27.643 ->  -865900
21:11:27.643 ->  +3500,-1650
21:11:27.643 ->  + 500,-1250 + 500,-1200 + 500,- 400 + 500,- 350
21:11:27.643 ->  + 500,- 350 + 500,-1250 + 500,- 350 + 500,- 400
21:11:27.676 ->  + 500,-1200 + 500,-1250 + 500,- 350 + 500,-1250
21:11:27.676 ->  + 500,- 350 + 500,- 400 + 500,-1200 + 550,-1200
21:11:27.676 ->  + 500
21:11:27.676 -> Sum: 26450
21:11:30.160 -> 
21:11:30.160 -> rawData[36]: 
21:11:30.160 ->  -2167400
21:11:30.160 ->  +3500,-1650
21:11:30.160 ->  + 500,-1250 + 500,-1200 + 500,- 400 + 500,- 350
21:11:30.160 ->  + 500,- 400 + 500,-1200 + 500,- 400 + 500,- 350
21:11:30.160 ->  + 500,-1200 + 500,-1250 + 500,- 350 + 500,-1250
21:11:30.160 ->  + 500,- 350 + 500,- 400 + 500,-1200 + 500,-1250
21:11:30.160 ->  + 500
21:11:30.160 -> Sum: 26450
21:11:31.187 -> 
21:11:31.187 -> rawData[36]: 
21:11:31.187 ->  -712800
21:11:31.187 ->  +3500,-1650
21:11:31.187 ->  + 500,-1250 + 450,-1250 + 500,- 400 + 500,- 350
21:11:31.187 ->  + 450,- 400 + 500,-1250 + 500,- 350 + 500,- 400
21:11:31.187 ->  + 500,-1200 + 500,-1250 + 500,- 350 + 500,-1250
21:11:31.187 ->  + 500,- 350 + 500,- 400 + 500,-1200 + 500,-1250
21:11:31.220 ->  + 500
21:11:31.220 -> Sum: 26450

```

ArminJo, 2024-03-15:
I am sorry, but your dump is useless 😞

chaseadam, 2024-03-15:
Yep, I figured that was the case... I tried with an Arduino Uno board. It is an 8 bit platform, so the tRawData is split differently, but still looks like it is incorrect:

```
10:30:54.353 -> Protocol=PulseDistance Raw-Data=0xCB23 144 bits LSB first
10:30:54.353 -> Send on a 8 bit platform with: 
10:30:54.353 ->     uint32_t tRawData[]={0x126CB23, 0x15080000, 0x3E4030, 0x0, 0xCB23};
10:30:54.353 ->     IrSender.sendPulseDistanceWidthFromArray(38, 3450, 1650, 500, 1250, 500, 400, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);
```

The rawData output appears to be similarly not useful...

```
10:35:28.568 -> rawData[36]: 
10:35:28.568 ->  -2259950
10:35:28.568 ->  +3450,-1650
10:35:28.568 ->  + 500,-1200 + 500,-1200 + 500,- 350 + 500,- 400
10:35:28.568 ->  + 500,- 350 + 500,-1200 + 500,- 350 + 500,- 350
10:35:28.568 ->  + 500,-1250 + 450,-1250 + 450,- 400 + 500,-1200
10:35:28.568 ->  + 500,- 350 + 500,- 400 + 450,-1250 + 600,-1100
10:35:28.601 ->  + 450
10:35:28.601 -> Sum: 26100
10:35:29.661 -> 
10:35:29.661 -> rawData[36]: 
10:35:29.661 ->  -756750
10:35:29.661 ->  +3500,-1600
10:35:29.661 ->  + 500,-1200 + 500,-1250 + 500,- 350 + 450,- 400
10:35:29.661 ->  + 500,- 350 + 500,-1200 + 500,- 350 + 550,- 350
10:35:29.661 ->  + 500,-1200 + 500,-1200 + 500,- 350 + 500,-1200
10:35:29.661 ->  + 500,- 400 + 450,- 400 + 500,-1200 + 500,-1200
10:35:29.694 ->  + 500
10:35:29.694 -> Sum: 26150
```

I assume I should be using the "ReceiveDemo" for this investigation. Unless you have specific suggestions, we may have reached the end of my rabbit hole.

chaseadam, 2024-03-15:
for completeness, this is the scope data on TSOP98638 data pin for the corresponding signals.
[DS0000.CSV](https://github.com/Arduino-IRremote/Arduino-IRremote/files/14617172/DS0000.CSV)

ArminJo, 2024-03-15:
It seems to be a important secret, that you do not tell how you created the raw data...
But this way it saves a few lines for posting 😞

chaseadam, 2024-03-15:
ah, my apologies, I believe this is what you are requesting:

```
1:37:07.485 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11:37:07.518 -> configsip: 0, SPIWP:0xee
11:37:07.518 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:37:07.518 -> mode:DIO, clock div:1
11:37:07.518 -> load:0x3fff0030,len:1184
11:37:07.518 -> load:0x40078000,len:13260
11:37:07.518 -> load:0x40080400,len:3028
11:37:07.518 -> entry 0x400805e4
11:37:07.650 -> START /tmp/arduino_modified_sketch_181257/ReceiveDemo.ino from Mar 14 2024
11:37:07.650 -> Using library version 4.3.0
11:37:07.683 -> Enabling IRin...
11:37:07.683 -> Ready to receive IR signals of protocols: NEC/NEC2/Onkyo/Apple, Panasonic/Kaseikyo, Denon/Sharp, Sony, RC5, RC6, LG, JVC, Samsung, FAST, Whynter, Lego Power Functions, Bosewave , MagiQuest, Universal Pulse Distance Width, Hash at pin 15
11:37:07.683 -> 
11:37:07.683 -> If you connect debug pin 16 to ground, raw data is always printed
11:37:07.683 -> 8000 us is the (minimum) gap, after which the start of a new IR packet is assumed
11:37:07.716 -> 20 us are subtracted from all marks and added to all spaces for decoding
11:37:13.383 -> 
11:37:13.383 -> Protocol=PulseWidth Raw-Data=0x0 144 bits LSB first
11:37:13.383 -> Send on a 32 bit platform with: 
11:37:13.383 ->     uint64_t tRawData[]={0x0, 0x40000000000, 0x0};
11:37:13.383 ->     IrSender.sendPulseDistanceWidthFromArray(38, 3550, 1650, 1350, 400, 500, 1250, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);
<shorted debug pin>
11:38:51.885 -> 
11:38:51.885 -> rawData[36]: 
11:38:51.885 ->  -3276750
11:38:51.885 ->  +3550,-1650
11:38:51.885 ->  + 550,-1200 + 500,-1200 + 500,- 350 + 550,- 350
11:38:51.885 ->  + 500,- 350 + 550,-1200 + 500,- 350 + 500,- 400
11:38:51.885 ->  + 500,-1200 + 550,-1200 + 500,- 350 + 550,-1200
11:38:51.885 ->  + 500,- 350 + 550,- 300 + 600,-1150 + 550,-1150
11:38:51.885 ->  + 550
11:38:51.885 -> Sum: 26500
```

ArminJo, 2024-03-15:
And now....
Connect pin 16 to ground!

chaseadam, 2024-03-15:
forgot to mark where I changed the pin 16 ground in that printout. Updated above and clarified below. It seems the debug mode is not presenting as much data as we expect.

I started up with pin 16 to ground which resulted in:

```
11:45:59.038 -> 
11:45:59.038 -> rawData[36]: 
11:45:59.038 ->  -3276750
11:45:59.038 ->  +3500,-1700
11:45:59.038 ->  + 450,-1250 + 500,-1250 + 450,- 400 + 500,- 350
11:45:59.038 ->  + 500,- 400 + 450,-1250 + 500,- 400 + 450,- 400
11:45:59.071 ->  + 500,-1250 + 500,-1200 + 500,- 400 + 450,-1250
11:45:59.071 ->  + 550,- 350 + 450,- 400 + 500,-1250 + 450,-1250
11:45:59.071 ->  + 500
11:45:59.071 -> Sum: 26450
11:46:01.822 -> 
11:46:01.822 -> Received noise or an unknown (or not yet enabled) protocol
11:46:01.822 -> rawData[36]: 
11:46:01.822 ->  -2459750
11:46:01.822 ->  +3550,-1600
11:46:01.822 ->  + 550,-1200 + 500,-1250 + 500,- 350 + 550,- 300
11:46:01.822 ->  + 550,- 350 + 500,-1200 + 550,- 350 + 550,- 300
11:46:01.855 ->  + 500,-1250 + 550,-1150 + 550,- 350 + 500,-1200
11:46:01.855 ->  + 550,- 350 + 500,- 350 + 550,-1200 + 650,-1050
11:46:01.855 ->  + 600
11:46:01.855 -> Sum: 26550
11:46:06.259 -> 
```

then removed from ground and got:

```
11:46:06.259 -> 
11:46:06.259 -> Protocol=PulseDistance Raw-Data=0xCB23 144 bits LSB first
11:46:06.259 -> Send on a 32 bit platform with: 
11:46:06.259 ->     uint64_t tRawData[]={0x150800000126CB23, 0x464030, 0xCB23};
11:46:06.292 ->     IrSender.sendPulseDistanceWidthFromArray(38, 3500, 1650, 450, 1250, 450, 400, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);
11:46:07.848 -> 
11:46:07.848 -> Protocol=PulseDistance Raw-Data=0xCB23 144 bits LSB first
11:46:07.848 -> Send on a 32 bit platform with: 
11:46:07.848 ->     uint64_t tRawData[]={0x150820000126CB23, 0x464030, 0xCB23};
11:46:07.848 ->     IrSender.sendPulseDistanceWidthFromArray(38, 3450, 1700, 450, 1250, 450, 400, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);
11:46:09.044 -> 
11:46:09.044 -> Protocol=PulseDistance Raw-Data=0xCB23 144 bits LSB first
11:46:09.044 -> Send on a 32 bit platform with: 
11:46:09.044 ->     uint64_t tRawData[]={0x150800000126CB23, 0x464030, 0xCB23};
11:46:09.044 ->     IrSender.sendPulseDistanceWidthFromArray(38, 3500, 1650, 450, 1250, 450, 400, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);
11:46:10.202 -> 
11:46:10.202 -> Protocol=PulseDistance Raw-Data=0xCB23 144 bits LSB first
11:46:10.202 -> Send on a 32 bit platform with: 
11:46:10.202 ->     uint64_t tRawData[]={0x150820000126CB23, 0x464030, 0xCB23};
11:46:10.202 ->     IrSender.sendPulseDistanceWidthFromArray(38, 3550, 1650, 450, 1250, 450, 400, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);
```

ArminJo, 2024-03-15:
Please do it again with pin 16 constantly connected and privide the complete log.

chaseadam, 2024-03-15:
here it is, 2x on/off sequences:

```
14:56:22.640 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:56:22.640 -> configsip: 0, SPIWP:0xee
14:56:22.640 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:56:22.640 -> mode:DIO, clock div:1
14:56:22.640 -> load:0x3fff0030,len:1184
14:56:22.640 -> load:0x40078000,len:13260
14:56:22.640 -> load:0x40080400,len:3028
14:56:22.640 -> entry 0x400805e4
14:56:22.839 -> START /tmp/arduino_modified_sketch_981185/ReceiveDemo.ino from Mar 15 2024
14:56:22.839 -> Using library version 4.3.0
14:56:22.839 -> Enabling IRin...
14:56:22.839 -> Ready to receive IR signals of protocols: NEC/NEC2/Onkyo/Apple, Panasonic/Kaseikyo, Denon/Sharp, Sony, RC5, RC6, LG, JVC, Samsung, FAST, Whynter, Lego Power Functions, Bosewave , MagiQuest, Universal Pulse Distance Width, Hash at pin 15
14:56:22.872 -> 
14:56:22.872 -> If you connect debug pin 16 to ground, raw data is always printed
14:56:22.872 -> 15000 us is the (minimum) gap, after which the start of a new IR packet is assumed
14:56:22.872 -> 20 us are subtracted from all marks and added to all spaces for decoding
14:56:24.958 -> 
14:56:24.958 -> Received noise or an unknown (or not yet enabled) protocol
14:56:24.958 -> rawData[36]: 
14:56:24.991 ->  -1961300
14:56:24.991 ->  +3500,-1650
14:56:24.991 ->  + 500,-1250 + 450,-1250 + 500,- 400 + 450,- 400
14:56:24.991 ->  + 450,- 400 + 500,-1250 + 450,- 400 + 500,- 350
14:56:24.991 ->  + 500,-1250 + 450,-1300 + 450,- 400 + 500,-1250
14:56:24.991 ->  + 500,- 350 + 500,- 400 + 500,-1200 + 550,-1200
14:56:24.991 ->  + 500
14:56:24.991 -> Sum: 26450
14:56:25.885 -> 
14:56:25.885 -> rawData[36]: 
14:56:25.885 ->  -600000
14:56:25.885 ->  +3500,-1700
14:56:25.885 ->  + 500,-1200 + 500,-1250 + 450,- 400 + 500,- 350
14:56:25.918 ->  + 500,- 400 + 500,-1250 + 450,- 400 + 450,- 400
14:56:25.918 ->  + 500,-1250 + 500,-1250 + 450,- 400 + 450,-1250
14:56:25.918 ->  + 550,- 350 + 500,- 400 + 450,-1250 + 500,-1250
14:56:25.918 ->  + 500
14:56:25.918 -> Sum: 26500
14:56:27.044 -> 
14:56:27.044 -> rawData[36]: 
14:56:27.044 ->  -832350
14:56:27.044 ->  +3500,-1700
14:56:27.077 ->  + 500,-1200 + 500,-1250 + 450,- 400 + 500,- 400
14:56:27.077 ->  + 500,- 350 + 500,-1250 + 500,- 350 + 500,- 350
14:56:27.077 ->  + 500,-1250 + 500,-1250 + 450,- 400 + 500,-1250
14:56:27.077 ->  + 500,- 350 + 500,- 350 + 500,-1250 + 500,-1200
14:56:27.077 ->  + 500
14:56:27.077 -> Sum: 26450
14:56:28.138 -> 
14:56:28.138 -> rawData[36]: 
14:56:28.138 ->  -764400
14:56:28.138 ->  +3450,-1700
14:56:28.171 ->  + 500,-1250 + 450,-1250 + 500,- 350 + 500,- 400
14:56:28.171 ->  + 500,- 350 + 500,-1250 + 450,- 400 + 500,- 400
14:56:28.171 ->  + 500,-1200 + 500,-1250 + 450,- 400 + 500,-1200
14:56:28.171 ->  + 500,- 400 + 500,- 350 + 500,-1250 + 500,-1200
14:56:28.171 ->  + 500
14:56:28.171 -> Sum: 26400
```

ArminJo, 2024-03-15:
Thanks, but I must give up. it makes no sense why 144 bits are received as 4 x 36 bit chunks with 700 ms pause between.

chaseadam, 2024-03-15:
no worries, thanks for engaging!

chaseadam, 2024-03-15:
to clarify, it outputs 4x 36 bit chunks for 4x 144bit messages, 36 bits for every message. I pressed the button 4 times:
> 2x on/off sequences

ArminJo, 2024-03-15:
can you modify ReceiveDemo
with:
```
            if (IrReceiver.decodedIRData.protocol == UNKNOWN || digitalRead(DEBUG_BUTTON_PIN) == LOW) {
                // We have debug enabled or an unknown protocol, print extended info
                if (IrReceiver.decodedIRData.protocol == UNKNOWN) {
                    Serial.println(F("Received noise or an unknown (or not yet enabled) protocol"));
                }
                IrReceiver.printIRResultRawFormatted(&Serial, true);
            }
            if (IrReceiver.decodedIRData.protocol != UNKNOWN) {
                /*
                 * The info output for a successful receive
                 */
                IrReceiver.printIRResultShort(&Serial);
                IrReceiver.printIRSendUsage(&Serial);
            }
```
and run again?

chaseadam, 2024-03-15:
I believe I found the raw display issue. I am no C coder, but looks like there is an issue with casting the length variable in IRrecv::printIRResultRawFormatted:
https://github.com/Arduino-IRremote/Arduino-IRremote/blob/02c0e24da9f75b03af53410ccfc02cf3380f23a8/src/IRReceive.hpp#L1476

From what I can tell, the *actual* rawlen value is 292 and number of bits is 144 based on output from the following modification to the ReceiverDemo:

```
                uint8_t tRawlen = IrReceiver.decodedIRData.rawlen;
                Serial.println(IrReceiver.decodedIRData.numberOfBits);
                Serial.println(tRawlen);
```

results in "36"
```
17:15:44.237 -> rawData[36]: 
17:15:44.237 ->  -3276750
17:15:44.237 ->  +3450,-1700
17:15:44.237 ->  + 450,-1300 + 400,-1300 + 550,- 350 + 400,- 450
17:15:44.237 ->  + 450,- 450 + 400,-1300 + 400,- 500 + 400,- 450
17:15:44.237 ->  + 400,-1300 + 450,-1300 + 450,- 400 + 450,-1300
17:15:44.270 ->  + 450,- 400 + 450,- 450 + 400,-1300 + 450,-1300
17:15:44.270 ->  + 400
17:15:44.270 -> Sum: 26350
17:15:44.270 -> 145
17:15:44.270 -> 36
```

vs 

```
                Serial.println(IrReceiver.decodedIRData.numberOfBits);
                Serial.println(IrReceiver.decodedIRData.rawlen);
```
results in "292"

```
17:13:25.077 -> rawData[36]: 
17:13:25.077 ->  -3276750
17:13:25.077 ->  +3450,-1700
17:13:25.077 ->  + 500,-1250 + 500,-1200 + 500,- 350 + 550,- 350
17:13:25.077 ->  + 400,- 450 + 500,-1250 + 400,- 450 + 450,- 400
17:13:25.110 ->  + 450,-1300 + 500,-1250 + 450,- 400 + 500,-1250
17:13:25.110 ->  + 400,- 450 + 500,- 350 + 450,-1300 + 500,-1200
17:13:25.110 ->  + 450
17:13:25.110 -> Sum: 26350
17:13:25.110 -> 144
17:13:25.110 -> 292
```

chaseadam, 2024-03-15:
> can you modify ReceiveDemo with:
>
> ```
>             if (IrReceiver.decodedIRData.protocol == UNKNOWN || digitalRead(DEBUG_BUTTON_PIN) == LOW) {
>                 // We have debug enabled or an unknown [...]

```
17:23:03.328 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:23:03.328 -> configsip: 0, SPIWP:0xee
17:23:03.328 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:23:03.328 -> mode:DIO, clock div:1
17:23:03.328 -> load:0x3fff0030,len:1184
17:23:03.328 -> load:0x40078000,len:13260
17:23:03.328 -> load:0x40080400,len:3028
17:23:03.328 -> entry 0x400805e4
17:23:03.494 -> START /tmp/arduino_modified_sketch_165244/ReceiveDemo.ino from Mar 15 2024
17:23:03.494 -> Using library version 4.3.0
17:23:03.494 -> Enabling IRin...
17:23:03.494 -> Ready to receive IR signals of protocols: Universal Pulse Distance Width, at pin 15
17:23:03.494 -> 
17:23:03.494 -> If you connect debug pin 16 to ground, raw data is always printed
17:23:03.494 -> 14000 us is the (minimum) gap, after which the start of a new IR packet is assumed
17:23:03.527 -> 20 us are subtracted from all marks and added to all spaces for decoding
17:23:07.273 -> 
17:23:07.273 -> rawData[36]: 
17:23:07.273 ->  -3276750
17:23:07.273 ->  +3500,-1650
17:23:07.273 ->  + 450,-1300 + 500,-1200 + 550,- 350 + 400,- 450
17:23:07.273 ->  + 400,- 500 + 450,-1250 + 500,- 350 + 450,- 450
17:23:07.306 ->  + 500,-1200 + 450,-1300 + 500,- 350 + 550,-1200
17:23:07.306 ->  + 500,- 350 + 450,- 450 + 500,-1200 + 500,-1250
17:23:07.306 ->  + 400
17:23:07.306 -> Sum: 26350
17:23:07.306 -> Protocol=PulseDistance Raw-Data=0xCB23 144 bits LSB first
17:23:07.306 -> Send on a 32 bit platform with: 
17:23:07.306 ->     uint64_t tRawData[]={0x150820000126CB23, 0x684030, 0xCB23};
17:23:07.306 ->     IrSender.sendPulseDistanceWidthFromArray(38, 3500, 1650, 450, 1250, 450, 400, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);
```

chaseadam, 2024-03-15:
aside: the compatibility across mixed-bit length platforms living inside of the arduino framework looks very challenging and with loads of "gotchas"

chaseadam, 2024-03-15:
I went digging and swapped the `uint8_t` for `uint32_t` and looks like a winner (not the right way).

```

17:29:37.573 -> rawData[292]: 
17:29:37.573 ->  -3276750
17:29:37.573 ->  +3500,-1650
17:29:37.573 ->  + 450,-1300 + 500,-1200 + 450,- 450 + 450,- 400
17:29:37.573 ->  + 450,- 450 + 500,-1200 + 500,- 350 + 550,- 350
17:29:37.573 ->  + 500,-1200 + 450,-1300 + 500,- 350 + 550,-1200
17:29:37.606 ->  + 400,- 450 + 450,- 450 + 500,-1200 + 450,-1300
17:29:37.606 ->  + 500,- 350 + 550,-1200 + 500,-1200 + 550,- 350
17:29:37.606 ->  + 400,- 450 + 500,-1250 + 450,- 400 + 500,- 350
17:29:37.606 ->  + 550,-1200 + 500,- 350 + 550,- 350 + 500,- 350
17:29:37.606 ->  + 550,- 350 + 500,- 350 + 500,- 400 + 500,- 350
17:29:37.606 ->  + 450,- 400 + 450,- 450 + 450,- 400 + 500,- 400
17:29:37.606 ->  + 400,- 450 + 400,- 400 + 600,- 350 + 400,- 450
17:29:37.606 ->  + 500,- 350 + 450,- 450 + 400,- 450 + 450,- 400
17:29:37.639 ->  + 550,- 350 + 500,- 350 + 500,- 400 + 400,- 450
17:29:37.639 ->  + 450,- 450 + 500,- 350 + 450,- 400 + 550,-1200
17:29:37.639 ->  + 500,- 350 + 550,- 350 + 500,- 350 + 550,- 350
17:29:37.639 ->  + 450,-1250 + 550,- 350 + 500,-1200 + 550,- 350
17:29:37.639 ->  + 500,-1200 + 450,- 450 + 500,- 350 + 400,- 450
17:29:37.639 ->  + 550,- 350 + 400,- 450 + 400,- 450 + 550,- 350
17:29:37.639 ->  + 500,-1200 + 550,-1200 + 450,- 400 + 450,- 450
17:29:37.639 ->  + 500,- 350 + 550,- 350 + 500,- 350 + 450,- 400
17:29:37.672 ->  + 450,- 450 + 500,- 350 + 450,-1300 + 500,- 350
17:29:37.672 ->  + 450,- 400 + 450,- 450 + 500,- 350 + 450,-1300
17:29:37.672 ->  + 450,- 400 + 550,-1200 + 450,-1250 + 550,- 350
17:29:37.672 ->  + 450,- 400 + 500,- 350 + 550,- 350 + 500,- 350
17:29:37.672 ->  + 450,- 450 + 400,- 450 + 400,- 450 + 450,- 450
17:29:37.672 ->  + 500,- 350 + 400,- 500 + 400,- 450 + 500,- 350
17:29:37.672 ->  + 450,- 450 + 400,- 450 + 500,- 350 + 450,- 450
17:29:37.672 ->  + 400,- 450 + 500,- 400 + 400,- 450 + 500,- 350
17:29:37.705 ->  + 450,- 450 + 450,- 400 + 450,- 400 + 450,- 450
17:29:37.705 ->  + 500,- 350 + 500,- 350 + 450,- 450 + 500,- 350
17:29:37.705 ->  + 450,- 450 + 400,- 450 + 500,- 350 + 550,- 350
17:29:37.705 ->  + 400,- 450 + 550,- 350 + 400,- 450 + 500,- 350
17:29:37.705 ->  + 450,- 450 + 500,- 350 + 400,- 500 + 500,- 350
17:29:37.705 ->  + 450,- 400 + 550,- 350 + 500,- 350 + 450,- 450
17:29:37.705 ->  + 500,- 350 + 500,- 350 + 450,- 450 + 450,- 400
17:29:37.738 ->  + 500,- 400 + 500,-1200 + 500,- 400 + 400,-1300
17:29:37.738 ->  + 500,- 350 + 450,- 450 + 500,- 350 + 450,- 450
17:29:37.738 ->  + 500
17:29:37.738 -> Sum: 151500
17:29:37.738 -> Protocol=PulseDistance Raw-Data=0xCB23 144 bits LSB first
17:29:37.738 -> Send on a 32 bit platform with: 
17:29:37.738 ->     uint64_t tRawData[]={0x150800000126CB23, 0x684030, 0xCB23};
17:29:37.738 ->     IrSender.sendPulseDistanceWidthFromArray(38, 3500, 1650, 500, 1250, 500, 400, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);
```

ArminJo, 2024-03-15:
Thanks for being so persistent. 🥇 I corrected the bug. There was a second place where overflow happend.

chaseadam, 2024-03-15:
hooray! now back to the original issue... it appears to be capturing the last 2 bytes (maybe more)  of the 144 bits incorrectly.

ArminJo, 2024-03-15:
I am working on it....

chaseadam, 2024-03-15:
ah! the overflow comment was more than just the rawlen thing. Thanks!

ArminJo, 2024-03-15:
No, I guess, there is another bug. still hunting...

ArminJo, 2024-03-16:
Ok, please get the latest version from this repo, it should work now.

chaseadam, 2024-03-18:
I haven't validated a crafted IR signal yet, but this is very promising:

```
08:52:14.641 -> rawData[292]: 
08:52:14.641 ->  -3276750
08:52:14.641 ->  +3500,-1650
08:52:14.674 ->  + 500,-1200 + 500,-1250 + 500,- 350 + 500,- 400
08:52:14.674 ->  + 500,- 350 + 500,-1250 + 500,- 350 + 500,- 400
08:52:14.674 ->  + 450,-1250 + 500,-1250 + 500,- 350 + 500,-1250
08:52:14.674 ->  + 450,- 400 + 500,- 350 + 500,-1250 + 500,-1200
08:52:14.674 ->  + 500,- 400 + 450,-1250 + 500,-1250 + 500,- 350
08:52:14.674 ->  + 500,- 400 + 450,-1250 + 500,- 350 + 500,- 400
08:52:14.674 ->  + 500,-1250 + 500,- 350 + 500,- 400 + 500,- 350
08:52:14.674 ->  + 500,- 350 + 500,- 350 + 550,- 350 + 500,- 400
08:52:14.707 ->  + 450,- 400 + 500,- 350 + 500,- 400 + 500,- 350
08:52:14.707 ->  + 500,- 400 + 450,- 400 + 450,- 400 + 500,- 400
08:52:14.707 ->  + 450,- 400 + 500,- 350 + 500,- 400 + 500,- 350
08:52:14.707 ->  + 500,- 400 + 500,-1200 + 550,- 350 + 450,- 400
08:52:14.707 ->  + 500,- 350 + 500,- 400 + 500,- 350 + 500,-1250
08:52:14.707 ->  + 500,- 350 + 500,- 350 + 500,- 400 + 450,- 400
08:52:14.707 ->  + 500,-1250 + 500,- 350 + 500,-1250 + 450,- 400
08:52:14.707 ->  + 500,-1250 + 500,- 350 + 500,- 400 + 450,- 400
08:52:14.740 ->  + 500,- 350 + 500,- 400 + 450,- 400 + 500,- 350
08:52:14.740 ->  + 500,-1250 + 500,-1200 + 500,- 400 + 500,- 350
08:52:14.740 ->  + 500,- 400 + 450,- 400 + 500,- 350 + 500,- 400
08:52:14.740 ->  + 450,- 400 + 500,- 350 + 550,-1200 + 500,- 350
08:52:14.740 ->  + 500,-1250 + 500,- 350 + 500,-1250 + 500,- 350
08:52:14.740 ->  + 500,-1250 + 500,-1200 + 500,- 400 + 500,- 350
08:52:14.740 ->  + 500,- 400 + 450,- 400 + 450,- 400 + 500,- 400
08:52:14.773 ->  + 500,- 350 + 500,- 350 + 500,- 400 + 500,- 350
08:52:14.773 ->  + 500,- 400 + 450,- 400 + 500,- 350 + 500,- 400
08:52:14.773 ->  + 450,- 400 + 500,- 350 + 500,- 400 + 500,- 350
08:52:14.773 ->  + 500,- 400 + 450,- 400 + 450,- 400 + 500,- 400
08:52:14.773 ->  + 500,- 350 + 500,- 400 + 450,- 400 + 450,- 400
08:52:14.773 ->  + 500,- 400 + 450,- 400 + 500,- 350 + 500,- 400
08:52:14.773 ->  + 500,- 350 + 500,- 350 + 500,- 400 + 500,- 350
08:52:14.773 ->  + 500,- 400 + 500,- 350 + 500,- 350 + 500,- 400
08:52:14.806 ->  + 500,- 350 + 500,- 350 + 500,- 400 + 500,- 350
08:52:14.806 ->  + 500,- 400 + 500,- 350 + 450,- 400 + 500,- 400
08:52:14.806 ->  + 500,- 350 + 500,- 350 + 500,- 400 + 500,- 350
08:52:14.806 ->  + 500,-1250 + 500,-1200 + 500,-1250 + 500,- 350
08:52:14.806 ->  + 500,-1250 + 500,-1200 + 500,-1250 + 500,-1250
08:52:14.806 ->  + 500
08:52:14.806 -> Sum: 157450
08:52:14.806 -> Protocol=PulseDistance Raw-Data=0xF700 144 bits LSB first
08:52:14.839 -> Send on a 32 bit platform with: 
08:52:14.839 ->     uint64_t tRawData[]={0x150820000126CB23, 0x354030, 0xF700};
08:52:14.839 ->     IrSender.sendPulseDistanceWidthFromArray(38, 3500, 1650, 500, 1250, 500, 400, &tRawData[0], 144, PROTOCOL_IS_LSB_FIRST, <RepeatPeriodMillis>, <numberOfRepeats>);
```

closed 2024-03-30 by ArminJo