Arduino-IRremote/Arduino-IRremote #1217: [Bug]: MagiQuest New Encoding Scheme

Siege36, 2024-04-02:
### The type of board you are using.

ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

esp32 dev kit

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

Other - please specify below

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [X] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [X] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [X] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [X] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. Enable Simple Receiver with MagiQuest
2. View output

### What happens?

I have 3 new MagiQuest wands that I got from Great Wolf Lodge during a very recent stay (4/1/24). After coming home and trying to decode the IR signals, it seems that there has been some updated.
It looks like MagiQuest has updated their encoding scheme. I have tried both with Simple Reciever and Demo Reciever and it only outputs raw IR recepits.
I have also tried the MagiQuest library in ESPHome to [similar results.](https://github.com/esphome/issues/issues/4048) I am unable to get any library to recognize what it is recieving is a MagiQuest reception. However both are outputing raw data logs.

### The serial output which indicates the error happened.

```shell
Received noise or an unknown (or not yet enabled) protocol
rawData[62]: 
 -3276750
 + 150,-2150
 + 150,-2150 + 200,-2100 + 150,-2150 + 500,-1800
 + 500,-1800 + 150,-2150 + 150,-1000 + 500,- 650
 + 150,-1000 + 500,- 650 + 500,-2950 + 550,-2900
 + 300,-2000 + 150,-1000 + 500,-2950 + 500,- 650
 + 450,-3000 + 150,-2150 + 150,-2150 + 450,- 700
 + 500,-2950 + 150,-2150 + 150,-2150 + 500,-1800
 + 500,- 650 + 500,- 650 + 450,-1850 + 150,-2150
 + 550,- 600 + 500
Sum: 63750

Received noise or an unknown (or not yet enabled) protocol
rawData[62]: 
 -449850
 + 200,-2100
 + 200,-2100 + 200,-2100 + 200,-2100 + 500,-1800
 + 500,-4100 + 250,- 950 + 500,- 650 + 150,-1000
 + 500,- 650 + 450,-1850 + 150,-1000 + 500,-2950
 + 150,-2150 + 200,- 950 + 500,-2950 + 450,- 700
 + 500,-2950 + 150,-2150 + 150,-2150 + 500,-1800
 + 150,-2150 + 150,-2150 + 150,-1000 + 550,- 600
 + 500,-1800 + 500,- 650 + 450,- 700 + 450,-1850
 + 150,-2150 + 150
Sum: 62300

Received noise or an unknown (or not yet enabled) protocol
rawData[58]: 
 -54050
 + 150,-1000
 + 200,- 950 + 200,-2100 + 150,-2150 + 250,-2050
 + 550,-1750 + 550,-1750 + 250,-2050 + 250,- 950
 + 500,-1800 + 500,- 650 + 450,-3000 + 500,-2950
 + 250,-2050 + 150,-1000 + 500,- 650 + 150,-2150
 + 450,- 700 + 500,-7550 + 500,- 650 + 500,-2950
 + 150,-2150 + 150,-2150 + 500,-1800 + 500,- 650
 + 500,- 650 + 500,-4100 + 450,- 700 + 500
Sum: 63800
```

### What were you expecting to happen?

Some sort of wandID

### Additional context.

I have 3 new wands and they're all behaving the same. I'd be happy to test further and provide more logs if needed.

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2024-04-03:
Hi this looks like a very compressed transmission with 2 different pulse length and 4 different pause length.
I am sorry but I guess nobody wants to decode this 😞.
Best is to ask the manufacurer for documentation first.

closed 2024-04-03 by ArminJo

Siege36, 2024-04-03:
@ArminJo is there a place I can list this as a feature request instead of just closing out the issue? I'd like to continue work on this and feel that although this is not a MagiQuest specific repo, it's probably the place with most viability.

ArminJo, 2024-04-03:
And who do you guess will be implementing this feature?
You are of course invited to to so 😀.