Arduino-IRremote/Arduino-IRremote #1226: [Bug]: `decode()` does not work when using `IrReceiver` inside a class.

nbourre, 2024-04-26:
### The type of board you are using.

Arduino ATmega2560 board (Mega)

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

26

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

Everything works fine if the `decode()` and everything are in the main file, but as soon as I move the code inside an `update()` method, the method `decode` don't work anymore. I want to encapsulate the remote inside a class to keep the main file as clean as possible.

What am I doing wrong?

Here the minimum code.

`Climatisation.h`

```ino
// Client class

#ifndef _CLIMATISATION_H_
#define _CLIMATISATION_H_

#define USE_IRREMOTE_HPP_AS_PLAIN_INCLUDE
#include <IRremote.hpp>

#include <Arduino.h>
#include <DHT.h>

class Climatisation {
public:
  // Constructeur
  Climatisation(int pinIR);

  // Méthodes
  void update();

private:
  int _pinIR = -1;
  bool _hasRemote = false;
};

#endif
```

`Climatisation.cpp`
```ino
// IMPORTANT!! Inclure IRremote.hpp avant les autres inclusions!
#include <IRremote.hpp>
#include "Climatisation.h"

Climatisation::Climatisation(int pinIR)
  : _pinIR(pinIR) {
  if (_pinIR > 0 ) {
    _hasRemote = true;
    IrReceiver.begin(_pinIR);
  }
}

void Climatisation::update() {
  if (_hasRemote) {
    if (IrReceiver.decode()) {
      Serial.println(IrReceiver.decodedIRData.decodedRawData, HEX);
      IrReceiver.resume();
    }
  }
}
```

`irremote_test.ino`
```ino
// Minimal test project which uses a class
// that includes an encapsulated IR remote control
#include "Climatisation.h"

Climatisation climatisation(26);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  climatisation.update();
}
```

### What happens?

The `decode()` never send `true` if encapsulated inside a class method.

### What were you expecting to happen?

`IrReceiver.decode()` should return `true` when the receiver get a valid ir code.

### Additional context.

I've tried the `IRremoteExtensionTest`, but it doesn't fully encapsulate the `IrReceiver` inside a class. The main file still needs to have a knowledge of `Irremote.hpp`

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2024-04-26:
> The main file still needs to have a knowledge of Irremote.hpp.

Please proof it!

nbourre, 2024-04-29:
The question is, why is `decode()` not working in the provided example?

ArminJo, 2024-04-29:
I updated the IRremoteExtensionTest example, and it works for me!

closed 2024-04-29 by ArminJo