Arduino-IRremote/Arduino-IRremote #1241: Liblary assigns pin wrongly

fawoorki, 2024-06-19:
### The type of board you are using.

ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

ESP32-WROOM-32S

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [X] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below

### The library version you are working with.

- [ ] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

This is the code that has issue (tiny sender works correctly):

#include <IRremote.h>

IRsend irsend(12);

void setup() {
  irsend.enableIROut(38);
}

void loop() {
  irsend.sendNEC(0x0, 0x7, 5);
}

### What happens?

When i specify and pin as a Irsend it always send everything on pin 0

### What were you expecting to happen?

Send IR signal on assigned pin

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2024-06-19:
Which pin does SimpleSender example use?

fawoorki, 2024-06-20:
> Which pin does SimpleSender example use?

i now use 23 but all works

ArminJo, 2024-06-20:
What happens, if you let pin 23 blink manually and attach a red LED?

fawoorki, 2024-06-20:
> What happens, if you let pin 23 blink manually and attach a red LED?

I didnt try making a led blink just powered it, and it was working normally, like every action that didnt include liblary is working fine with every pin.

ArminJo, 2024-06-20:
And what do you expect now?

fawoorki, 2024-06-20:
> And what do you expect now?

I just want to know if it's a problem with the library and eventually report a bug, or am I doing something wrong?

ArminJo, 2024-06-20:
And how can I help you?

fawoorki, 2024-06-20:
I just showed you part of my code that is responsible for sending the IR signal, and like I said, when I assign the pin in this case 12 as an IRsend, I get the output on the wrong pin (no matter what pin I assign, I have output always on pin 0). I want to know if I'm doing something wrong or if there is a problem with the library?

ArminJo, 2024-06-20:
And how can I decide it? It runs perfectly on all 12 boards I tested!