Arduino-IRremote/Arduino-IRremote #1255: [Bug]: zero bits received with NEC encoding on Arduino Uno R3

nikcain, 2024-08-02:
### The type of board you are using.

Arduino ATmega328* board (Uno, Nano, etc.)

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

Uno R3

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

8

### Example(s) you have checked while hunting the bug.

- [X] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

`#define DECODE_NEC 

#include "Arduino.h"
#include "IRremote.hpp"

IRrecv irrecv(8);

void setup() {
  Serial.begin(9600);
  Serial.println("lets go!");
  irrecv.enableIRIn();
}

void loop() {
  while (irrecv.decode()) {
    Serial.println(irrecv.decodedIRData.command);
    irrecv.printIRResultShort(&Serial);
    delay(200);
    irrecv.resume();
  }
}
`

I'm using an LG TV remote, and IRremote with DEBUG defined. 

### What happens?

The NEC protocol isn't recognised, and debug output says 0 bits.

### The serial output which indicates the error happened.

```shell
0
Protocol=UNKNOWN 0 bits (incl. gap and start) received
0
Protocol=UNKNOWN 0 bits (incl. gap and start) received
```

### What were you expecting to happen?

With 4.3.2 I get this:

21
Protocol=NEC Address=0x4 Command=0x15 Raw-Data=0xEA15FB04 32 bits LSB first

### Additional context.

I had a working application with IRRemote 4.3.2  I upgraded the library and it no longer detected the key presses on the remote. However the example code still worked, so I'm assuming this is my error, but I can't see how my code is that different from the samples. I stripped my code right back to just the way I was using irremote, and the problem persisted. What did I miss out that somehow still worked in 4.3.2?

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2024-08-03:
> [Verfolgungsbild][https://github.com/notifications/beacon/AA5TXHAGHJ7TMVINREAKFV3ZPPEDVA5CNFSM6AAAAABL47ZXU6WGG33NNVSW45C7OR4XAZNFJFZXG5LFVJRW63LNMVXHIX3JMTHJDQ5OSI.gif]

I can check, but it will take 2 weeks.

nikcain, 2024-08-03:
4.3.2 works for me, so I'm in no rush.

If there are any tests I can run that would help, just let me know

closed 2024-08-12 by ArminJo

ArminJo, 2024-08-12:
Now you get a depecated warning, because you use your own IRrecv instance / object.

commit ac291f90 (2024-08-12): Closes #1255
commit 7a115e8d (2024-08-12): Closes #1255