Arduino-IRremote/Arduino-IRremote #1260: [Bug]: Bumped version to 4.4.1 works without error if "timerStart(s50usTimer);" is replaced with "timerRestart(s50usTimer);"

mboehmerm, 2024-08-16:
### The type of board you are using.

ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

Nologo ESP32C3 Super Mini

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 10

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [X] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [X] Other - please specify below

### The library version you are working with.

- [ ] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

I tested the Bumped version to 4.4.1 with Arduino IDE, esp32 Core 3.0.4 and an ESP32-C3 Super Mini.
I used IRremote version 4.4.0 and IRTimer.hpp from the "Bumped version to 4.4.1".

It works for me without error if i replace in IRTimer.hpp
```
timerStart(s50usTimer);" 
```
with
```
timerRestart(s50usTimer);" 
```
Otherwise i get the error message : "E (4118) gptimer: gptimer_start(348): timer is not enabled yet"

Test program
```
#include "IRremote.hpp"
const int irReceiverPin = 10;

void setup() {
  Serial.begin(115200);
  delay(4000);

  IrReceiver.begin(irReceiverPin , ENABLE_LED_FEEDBACK);  
 
  Serial.println("IRremote Receive Test !"); 
}

void loop(void)
{
      if (IrReceiver.decode()) {
        IrReceiver.printIRResultShort(&Serial);
        IrReceiver.resume();
      }
}
```
Otherwise i get the error message : "E (4118) gptimer: gptimer_start(348): timer is not enabled yet"

### What happens?

It works for me without error if i replace in IRTimer.hpp
```
timerStart(s50usTimer);" 
```
with
```
timerRestart(s50usTimer);" 
```
Otherwise i get the error message : "E (4118) gptimer: gptimer_start(348): timer is not enabled yet"

### The serial output which indicates the error happened.

```shell
E (4118) gptimer: gptimer_start(348): timer is not enabled yet
```

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

akellai, 2024-08-19:
Your sample code works fine for me (no need to replace timerStart; same board - Nologo ESP32C3 Super Mini; esp32 by Espressif Systems 3.0.4). Here is the serial output:
IRremote Receive Test !
Protocol=NEC Address=0xBF00 Command=0x15 Raw-Data=0xEA15BF00 32 bits LSB first
Protocol=NEC Address=0xBF00 Command=0x15 Repeat gap=40950us

mboehmerm, 2024-08-20:
> Your sample code works fine for me (commit [39136b6](https://github.com/Arduino-IRremote/Arduino-IRremote/commit/39136b65a295f992c2c25f8b251ba839ba93ef3d); no need to replace timerStart; same board - Nologo ESP32C3 Super Mini; esp32 by Espressif Systems 3.0.4). Here is the serial output: IRremote Receive Test ! Protocol=NEC Address=0xBF00 Command=0x15 Raw-Data=0xEA15BF00 32 bits LSB first Protocol=NEC Address=0xBF00 Command=0x15 Repeat gap=40950us

[image]

akellai, 2024-08-20:
I was fairly certain I didn’t encounter your error, but after uninstalling and reinstalling the library, I see it now. After some Googling, it appears that the error message is misleading and doesn’t actually prevent the timer from working. I applied the suggested fix, and you’re right—the message is gone. Thanks!

ArminJo, 2024-08-21:
Issue is fixed now.

closed 2024-08-21 by ArminJo