Arduino-IRremote/Arduino-IRremote #1262: problem with  RC6 protocol 36 bits

janvanaal, 2024-08-28:
### The type of board you are using.

Arduino ATmega328* board (Uno, Nano, etc.)

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

Arduino nano

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

RC5, RC6

### Pin(s) used for IR-receive, if not default.

Arduino pin number: default 3

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [X] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [X] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [X] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [X] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

I have been trying to program an Arduino with the IR RC6 protocol for quite some time now.

I want to use this to skip the commercials on television. I just can't seem to get it to work.

When I read the original remote control I see that it is 36 bits

and when I send it with the Arduino it is 20 bits.

Am I doing something wrong or is this not yet provided for?

I hope you will answer this.

Kind regards from the Netherlands, Jan van Aal.

### What happens?

the difference is 36 bits or 20 bits. My TV receiver does not respond to this.

### The serial output which indicates the error happened.

```shell
this is from the original remote sender:
Protocol=RC6 Address=0x26 Command=0x10 Toggle=1 Raw-Data=0x8087A610 36 bits MSB first
Send with: IrSender.sendRC6(0x26, 0x10, <numberOfRepeats>);

And this is what I get when I produce it with the arduino:
Protocol=RC6 Address=0x26 Command=0x11 Raw-Data=0x2611 20 bits MSB first
Send with: IrSender.sendRC6(0x26, 0x11, <numberOfRepeats>);

the difference is 36 bits or 20 bits. My TV receiver does not respond to this.

Thanks,
Jan van Aal

jhcvaal@kabelfoon.nl
```

### Final checklist for the bug report.

- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] I have browsed the examples for one, that matches my use case.
- [X] The title of the issue is helpful and relevant.

ArminJo, 2024-08-28:
Can you test the latest repository version?
And can you attach the raw timing like:

```
Send RC6A with 14 bit 0x2711 as extra
Protocol=RC6A Address=0xF1 Command=0x76 Extra=0x2711 Toggle=1 Raw-Data=0xA711F176 35 bits MSB first
Send with: IrSender.sendRC6A(0xF1, 0x76, <numberOfRepeats>, 0x2711);
rawData[58]:
 -1032900
 +2650,- 900
 + 450,- 450 + 450,- 450 + 450,- 850 +1400,-1300
 + 950,- 850 + 500,- 400 + 950,- 400 + 500,- 450
 + 450,- 850 + 500,- 400 + 500,- 400 + 950,- 850
 + 500,- 400 + 500,- 400 + 950,- 450 + 450,- 450
 + 450,- 450 + 450,- 400 + 500,- 850 + 500,- 400
 + 500,- 400 + 950,- 850 + 950,- 400 + 500,- 400
 + 500,- 850 + 950,- 450 + 450,- 850 + 500
Sum: 36850

```

janvanaal, 2024-08-28:
Hi Armin,

It's great that you want to help me. I'm home now, but I'll take care of that tomorrow

Greet, Jan van Aal.

closed 2024-09-09 by ArminJo