Arduino-IRremote/Arduino-IRremote #1272: sendSharp (address, command, repeats) doesn't work, but fix found

ArminJo, 2024-11-28:
### Discussed in https://github.com/Arduino-IRremote/Arduino-IRremote/discussions/1271

<div type='discussions-op-text'>

<sup>Originally posted by **gbhug5a** November 28, 2024</sup>
My Sharp TV doesn't respond to the sendSharp (address, command, repeats) function, but it works fine if I change the value in line 127 of ir_Denon.hpp from 0x200 to 0x100.  This is the 2-bit "Sharp" indicator in the Denon format.  Apparently zero is reserved for Denon, and 0x100 works for my TV.  I don't know if other values (0x200 or 0x300) work for other Sharp products.

In addition, my Sharp remote sends the code three times instead of two - normal, then inverted, then normal again.  So to have the function exactly duplicate what the remote sends, I inserted these two lines after line 141:

delay(DENON_AUTO_REPEAT_DISTANCE / MICROS_IN_ONE_MILLI);
sendPulseDistanceWidthData(&DenonProtocolConstants, tData, DENON_BITS);
 
One other thing I noticed, but didn't test, is that it appears the parameters order in this function at line 276 is wrong:

/*
 * Old function without parameter aNumberOfRepeats
 */
void IRsend::sendSharp(uint16_t aAddress, uint16_t aCommand) {
    sendDenon(aAddress, aCommand, **true, 0**);
}

My TV is model LC-32LB-1504, and the remote is model LC-RC1-14.
The library is v4.4.1.

</div>

ArminJo, 2024-11-28:
@gbhug5a **Thank you very much for testing sendSharp.** 👍 

I adjusted the code, such that 0x100 is default and for 0x200 you must call sendSharp2().

Before I release it, can you be so kind and test, what the non inverted/inverted sequence for a single and multiple repeats is.
And maybe also the length of the gaps, if they are not standard.

gbhug5a, 2024-11-28:
I'm not sure I understand your request.  The three-time transfer (normal/inverted/normal) is what the Sharp remote does for even the briefest keypress.  That's the minimum transmission.  So are you asking what happens if the key is held down?

The Sharp remote break between the three phases is about 45000-46000.  Yours is about 40700, but works fine.

ArminJo, 2024-11-28:
Sorry for being unspecific.

> So are you asking what happens if the key is held down?

Yes exactly, that is what i want to know 😀.

gbhug5a, 2024-11-28:
When the key is held down, the remote continues to send alternating inverted/normal packets with the same 45000 break between them. Here's the volume up key held down for a couple seconds:

   On    Off     On    Off     On    Off     On    Off

   282,  1810,   284,   758,   284,   758,   284,   758,
   284,   758,   284,   758,   284,   758,   284,  1808,
   287,   755,   287,  1806,   288,   754,   288,   754,
   288,   754,   288,  1805,   279,   762,   280, 47317,
   278,  1814,   280,   762,   280,   762,   280,   762,
   280,   762,   280,  1812,   282,  1811,   284,   758,
   284,  1808,   286,   756,   286,  1806,   289,  1804,
   279,  1813,   282,   760,   282,  1809,   285, 43109,
   287,  1804,   280,   762,   280,   762,   280,   762,
   280,   762,   280,   762,   280,   762,   280,  1812,
   282,   760,   282,  1810,   284,   758,   284,   758,
   284,   758,   284,  1808,   286,   756,   286, 47171,
 
   341,  1751,   281,   761,   281,   761,   281,   761,
   280,   762,   280,  1812,   283,  1809,   285,   757,
   285,  1808,   286,   756,   286,  1806,   289,  1803,
   281,  1811,   283,   759,   283,  1809,   285, 42970,
   280,  1812,   282,   760,   282,   759,   283,   759,
   282,   760,   282,   760,   282,   760,   282,  1810,
   284,   758,   284,  1808,   287,   755,   287,   755,
   287,   755,   287,  1805,   289,   752,   280, 47175,
 
   283,  1809,   286,   756,   286,   755,   287,   755,
   287,   755,   287,  1805,   289,  1803,   281,   761,
   281,  1811,   283,   759,   283,  1810,   284,  1808,
   286,  1806,   289,   753,   288,  1803,   281, 42972,
   288,  1804,   279,   763,   279,   763,   279,   763,
   279,   763,   279,   763,   279,   763,   279,  1813,
   281,   761,   281,  1811,   283,   759,   283,   759,
   283,   758,   284,  1808,   286,   755,   288, 47167,
 
   281,  1810,   284,   758,   284,   758,   285,   757,
   285,   757,   285,  1807,   287,  1805,   279,   763,
   279,  1813,   281,   761,   281,  1811,   283,  1809,
   285,  1807,   288,   754,   288,  1803,   281, 42972,
   279,  1813,   281,   760,   282,   760,   282,   760,
   282,   760,   282,   760,   282,   760,   282,  1810,
   284,   758,   284,  1808,   287,   755,   287, 617839

ArminJo, 2024-11-28:
Thanks!
I updated ir_Denon.hpp
Two questions left.
1. Is a command with only 1 normal/inverted accepted by the TV?
2. Is a command with 2 normal/inverted accepted as one press?
This would simplify programming and backward compatibility 😀

closed 2024-11-28 by ArminJo

gbhug5a, 2024-11-28:
The answer to your first question is yes, the TV does accept it.  But I found it to be not as reliable.  Sometimes I press a key and it doesn't take.  That may be because there is no long start bit in Sharp.  In any case, the normal/inverted/normal protocol seems to work best.

On the second question, I don't really have a way to test that.  If I do any version of 1 repeat, the TV only recognizes the first keypress.  But I suspect that's because it is coded not to do anything with a repeated key until it has been repeating for some time, like half a second.  But I will try to test more with the volume up key to see if I can find anything definitive.

To follow the behavior of the remote, it looks like the code would be this:

    // Data
    sendPulseDistanceWidthData(&DenonProtocolConstants, tData, DENON_BITS);

    while (tNumberOfCommands > 0) {

        // Inverted autorepeat frame
        delay(DENON_AUTO_REPEAT_DISTANCE / MICROS_IN_ONE_MILLI);
        sendPulseDistanceWidthData(&DenonProtocolConstants, tInvertedData, DENON_BITS);

        delay(DENON_AUTO_REPEAT_DISTANCE / MICROS_IN_ONE_MILLI);
        sendPulseDistanceWidthData(&DenonProtocolConstants, tData, DENON_BITS);

        tNumberOfCommands--;

/*        // skip last delay!
        if (tNumberOfCommands > 0) {
            // send repeated command with a fixed space gap
            delay( DENON_AUTO_REPEAT_DISTANCE / MICROS_IN_ONE_MILLI);
        }
*/
    }

ArminJo, 2024-11-28:
Thanks, but I am quite reluctant to change the code, since is must be backwards compatible and maybe other remotes work differently.
I added a comment "Maybe minimal number of repeats is 1 for a press to be reliable detected by some devices"  to help users.

gbhug5a, 2024-11-28:
In your new version, shouldn't the last parameter in this be "2" instead of "true"?

void IRsend::sendSharp2(uint8_t aAddress, uint8_t aCommand, int_fast8_t aNumberOfRepeats) {
    sendDenon(aAddress, aCommand, aNumberOfRepeats, **true**);
}

ArminJo, 2024-11-28:
It seems to late for me...
Thank you for being so attentive 👍

ArminJo, 2024-12-06:
I just got a Denon Remote and documented and changed the code to send 3 + n*2 repeat frames 😀

commit 95d0bfc6 (2024-11-28): fixes #1272 Sharp frame marker
commit 424d757f (2024-11-28): fixes #1272 Sharp frame marker
commit 403d3ee2 (2024-11-28): fixes #1272 Sharp frame marker