Arduino-IRremote/Arduino-IRremote #1295: [Bug]: When using IRremote and Adafruit_NeoPixel together, the app crashes

Nourbakhsh-Rad, 2025-04-27:
### The type of board you are using.

Other - please specify below

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

ESP32 Devkitc V4

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: IO4

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [x] Other - please specify below
- [x] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [ ] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [x] Other - please specify below

### The library version you are working with.

- [x] I use the [latest Arduino library version](../releases) and verified this!
- [ ] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

1. Using IRremote and Adafruit_NeoPixel to change LED color and brightness
2. IRremote pin = IO4
3. WS2812 LED pin = IO0

### What happens?

After the board is turned on, as soon as the LED lights up (Adafruit_NeoPixel::show() ), the board will restart

### The serial output which indicates the error happened.

```shell

```

### What were you expecting to happen?

**Decoding stack results**
0x40082b90: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 54
0x4008f6e1: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c line 87
0x40095b09: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 80
0x40092642: vTaskPriorityDisinheritAfterTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c line 5267
0x4008ff57: xQueueSemaphoreTake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c line 1842
0x400eb7ba: espShow at d:\Sketchbook\Arduino\libraries\Adafruit_NeoPixel\esp.c line 52
0x400eb6ff: Adafruit_NeoPixel::show() at d:\Sketchbook\Arduino\libraries\Adafruit_NeoPixel\Adafruit_NeoPixel.cpp line 3176
0x400e5cd3: OneLEDpx_state(LEDpx_State, unsigned char, LEDpx_color_t, unsigned char, LEDpx_color_t, unsigned char) at d:\Sketchbook\Arduino\libraries\SHNlibs\LPXutil.cpp line 509
0x400e452a: Initial() at F:\Electronic\SAAT co!\__2025__\T24M22h/T24M22h.ino line 413
0x400e454b: setup() at F:\Electronic\SAAT co!\__2025__\T24M22h/T24M22h.ino line 56
0x400f6ec7: loopTask(void*) at C:\Users\SAATco\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\main.cpp line 59
0x40090366: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 139

### Final checklist for the bug report.

- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] I have browsed the examples for one, that matches my use case.
- [x] The title of the issue is helpful and relevant.

ArminJo, 2025-04-27:
Why do you think, this is an issue in IRremote and not in ESP32 core?

Nourbakhsh-Rad, 2025-04-28:
1. IRremote and Adafruit_NeoPixel work fine on their own
2. This problem occurs when using ESP32 Arduino core 3.2.0 and does not exist in lower versions

The problem may be with the ESP32 Arduino core or Adafruit_NeoPixel  library....

closed 2025-04-28 by ArminJo