Arduino-IRremote/Arduino-IRremote #1301: [Bug]: no matching function for call to printirresultshort

fungreenfox, 2025-05-26:
### The type of board you are using.

ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266

### The boards name or FQBN (e.g. esp8266:esp8266:d1_mini:eesz=4M3M,xtal=80) for unspecified board.

4

### The board manager URL used for unspecified board.

https://

### What IDE are you using?

Arduino IDE

### What IR protocol are you using?

NEC

### Pin(s) used for IR-receive, if not default.

Arduino pin number: 35

### Example(s) you have checked while hunting the bug.

- [ ] SimpleReceiver
- [x] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below
- [x] I checked, if at least one of the examples was working.

### Example(s) to reproduce the issue.

- [ ] SimpleReceiver
- [x] ReceiveDemo
- [ ] SendRawDemo
- [ ] ReceiverTimingAnalysis
- [ ] TinyReceiver
- [ ] TinySender
- [ ] ReceiveAndSend
- [ ] SimpleSender
- [ ] SendDemo
- [ ] SendLGAirConditionerDemo
- [ ] UnitTest
- [ ] Other - please specify below

### The library version you are working with.

- [x] I use the [latest Arduino library version](../releases) and verified this!
- [x] I use the latest repo version ([download link](../archive/master.zip)) and verified this!

### What are the steps to reproduce this issue?

Version 441 works to the point of saying its ready to receive at pin35
version 442 at first update from 441, gives:

In file included from O:\Programmer\Arduino\arduino_radio7_Si4703_V1_Tiny_RTC_copy42.ino\IRremoteExtensionClass.h:35,
                 from O:\Programmer\Arduino\arduino_radio7_Si4703_V1_Tiny_RTC_copy42.ino\IRremoteExtensionClass.cpp:47:
o:\Programmer\Arduino\libraries\IRremote\src/IRremote.hpp:200:2: warning: #warning INFO: For ESP32, RP2040, mbed and particle boards SEND_PWM_BY_TIMER is enabled by default, since we have the resources and timing is more exact than the software generated one. If this is not intended, deactivate the line in IRremote.hpp over this warning message in file IRremote.hpp. [-Wcpp]
  200 | #warning INFO: For ESP32, RP2040, mbed and particle boards SEND_PWM_BY_TIMER is enabled by default, since we have the resources and timing is more exact than the software generated one. If this is not intended, deactivate the line in IRremote.hpp over this warning message in file IRremote.hpp.
      |  ^~~~~~~
O:\Programmer\Arduino\arduino_radio7_Si4703_V1_Tiny_RTC_copy42.ino\IRremoteExtensionClass.cpp: In member function 'bool IRExtensionClass::printIRResultShort(Print*, bool, bool)':
O:\Programmer\Arduino\arduino_radio7_Si4703_V1_Tiny_RTC_copy42.ino\IRremoteExtensionClass.cpp:57:44: error: no matching function for call to 'IRrecv::printIRResultShort(Print*&, bool&, bool&)'
   57 |     return MyIrReceiver->printIRResultShort(aSerial, aPrintRepeatGap, aCheckForRecordGapsMicros);
      |            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from o:\Programmer\Arduino\libraries\IRremote\src/IRremote.hpp:270:
o:\Programmer\Arduino\libraries\IRremote\src/IRremoteInt.h:264:10: note: candidate: 'bool IRrecv::printIRResultShort(Print*, bool)'
  264 |     bool printIRResultShort(Print *aSerial, bool aCheckForRecordGapsMicros = true);
      |          ^~~~~~~~~~~~~~~~~~
o:\Programmer\Arduino\libraries\IRremote\src/IRremoteInt.h:264:10: note:   candidate expects 2 arguments, 3 provided

exit status 1

Compilation error: no matching function for call to 'IRrecv::printIRResultShort(Print*&, bool&, bool&)'

### What happens?

does not compite with exit 1

### The serial output which indicates the error happened.

```shell
does not compite with exit 1
```

### What were you expecting to happen?

expecting no errors mentioned going from 441 to 442

### Final checklist for the bug report.

- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] I have browsed the examples for one, that matches my use case.
- [x] The title of the issue is helpful and relevant.

closed 2025-05-26 by ArminJo

ArminJo, 2025-05-27:
It works for me with the latest IRReceive.hpp and IRremoteInt.h.

fungreenfox, 2025-05-27:
Okay. I have just reverted to parent version. NP.